vpaeder/pymcp2221 #5: _read_flash: I think the data length is 2 bytes shorter than currently reported

ianccole, 2023-02-28:
Sorry to report another but...

https://github.com/vpaeder/pymcp2221/blob/e4c66973052516822bdf6d0de362c372a51dc207/mcp2221/__init__.py#L319

datasheet http://ww1.microchip.com/downloads/en/devicedoc/20005292c.pdf table 3-9 indicates that the length at byte index 2 is the number of bytes +2

So I see spurious characters at the end of the strings - but these are fixed if I subtract 2 from the length:

        data = self._write(0xb0, code)
        len = data[2] - 2
        return data[4:(4+len)]

vpaeder, 2023-03-01:
Thanks for mentioning this. Turns out data length for flash reads is defined in Table 3-4. The issue is specific to USB descriptors. Tables 3-7 to 3-9 specify that the 2 last bytes can be dropped only if the utf-16-encoded string is shorter than 60 bytes. I fixed the code to handle that and the fringe case correctly.

closed 2023-03-01 by vpaeder

ianccole, 2023-03-01:
Appreciate it - thanks