vpaeder/pymcp2221 #6: Device not connected?

enok71, 2023-05-19:
I'm trying to run the examples, but just keep getting "Device not connected" errors. E.g. this simple code:

`import mcp2221

devices=mcp2221.find_devices()
device=devices[0]
print(device)

m=mcp2221.MCP2221(device)
print(m.gpio0_read_function())
`
result:
`{'path': b'3-5.3.1:1.2', 'vendor_id': 1240, 'product_id': 221, 'serial_number': '', 'release_number': 256, 'manufacturer_string': '', 'product_string': '', 'usage_page': 0, 'usage': 0, 'interface_number': 2}
Traceback (most recent call last):
  File "/home/dinbox/Dinbox/tbb/./flash2221.py", line 10, in <module>
    print(m.gpio0_read_function())
  File "/home/dinbox/Dinbox/tbb/venv/lib/python3.10/site-packages/mcp2221/__init__.py", line 1284, in gpio0_read_function
    ret = self._gpio_read_function(0, mem)
  File "/home/dinbox/Dinbox/tbb/venv/lib/python3.10/site-packages/mcp2221/__init__.py", line 1248, in _gpio_read_function
    return self._read_sram_byte(SramDataSubcode.GPSettings, gpio_num, [0, 1, 2])
  File "/home/dinbox/Dinbox/tbb/venv/lib/python3.10/site-packages/mcp2221/__init__.py", line 429, in _read_sram_byte
    ret = self._read_sram(code)
  File "/home/dinbox/Dinbox/tbb/venv/lib/python3.10/site-packages/mcp2221/__init__.py", line 331, in _read_sram
    data = self._write(0x61)
  File "/home/dinbox/Dinbox/tbb/venv/lib/python3.10/site-packages/mcp2221/__init__.py", line 235, in _write
    raise IOException("Device not connected.")
mcp2221.exceptions.IOException: Device not connected.
`

What am I doing wrong? Obviously it is connected.

enok71, 2023-05-19:
It turns out I can run the above code successfully as root. But why not as the ordinary user who has write access to both the serial device /dev/ttyACM0 and the i2c device /dev/i2c-17 in my case?

vpaeder, 2023-05-19:
I'd say it's related to hidapi, for which you need specific udev rules (see [here](https://github.com/trezor/cython-hidapi#id6)).

enok71, 2023-05-19:
Ok thanks!

enok71, 2023-05-22:
I can confirm that udev rules work well. I used the following rules to set up permissions:

```
UBSYSTEM=="usb", ATTRS{idVendor}=="04d8", ATTRS{idProduct}=="00dd", MODE="0660", GROUP="plugdev", TAG+="uaccess", TAG+="udev-acl"
KERNEL=="hidraw*", ATTRS{idVendor}=="04d8", ATTRS{idProduct}=="00dd", MODE="0660", GROUP="plugdev", TAG+="uaccess", TAG+="udev-acl"
# stop the ModemManager from trying to use this port as a modem!
ATTRS{idVendor}=="04d8", ATTRS{idProduct}=="00dd", ENV{ID_MM_DEVICE_IGNORE}="1"
```

Maybe an error message in pymcp2221 mentioning permissions failure instead of "device not connected" would still be helpful though, if it can be achieved?

vpaeder, 2023-05-22:
Ok nice.
Yes that's indeed a better behaviour. I fixed the *open* method to throw an IOException if it fails.

closed 2023-05-30 by vpaeder