vpaeder/pymcp2221 #11: Question, length of Product and Manufacturer strings

harkwoodservices, 2023-09-02:
This library is exactly what I'm looking for, so I can update the MCP2221A details without needing the Windows .DLL's, great work.

However, I am finding that when trying to use write_usb_product_descriptor and write_usb_manufacturer_descriptor the whole string is not always updated.

If I use write_usb_product_descriptor("12345678901234567890123456789") it works fine

If I use write_usb_product_descriptor("Product-Name Version 2"), what I get written is "Product-Name Version", it is always truncated after the 20th character.

I'm probably missing something simple, any pointers are welcome.

vpaeder, 2023-09-02:
This is because strings shorter than 29 characters must be terminated (with \0). I changed the code to do that automatically.

harkwoodservices, 2023-09-02:
That would explain it. Thanks for the quick fix.

closed 2023-09-02 by harkwoodservices